UIoT/middleware_sc_execl #4: Testar e homologar todos os equipamentos do Laborátorio

felipeduarte34, 2016-04-16:
Deve ser feito testes e implementados codigos de exemplo e tutoriais de todos os sensores e microcontroladores do laboratorio. Com o intuito de homologar todos os equipamentos para desenvolvimento de produtos para o projeto.

Bunoviske, 2016-04-21:
PIR SENSOR (SENSOR DE PRESENÇA)

Sensor que detecta a presença de objetos ou pessoas. Quando detecta movimento, ele manda um sinal digital HIGH no pino de saída durante 3 segundos no mínimo. Ele é eficiente em perceber movimento, porém fica inativo durante o tempo que está mandando sinal HIGH, representando um ponto negativo dependendo da aplicação.

Código do sensor quando foi utilizado como contador de pessoas: middleware_sc_execl/execl_use_cases_codes/new_codes/arduino/presence_sensor/person_counter_presence_sensor_code

Bunoviske, 2016-04-21:
HC-SR04 (SENSOR ULTRASÔNICO)

Sensor que calcula a distância de objetos a sua frente por meio de ondas ultrasônicas. Muito preciso a curtas distâncias (menos que 50cm), e muito impreciso a longas distâncias, pois há interferência do ambiente na hora da leitura dos dados.

Código em que o sensor foi usado como contador de pessoas: middleware_sc_execl/execl_use_cases_codes/new_codes/arduino/super_sonic_presence_sensor/person_counter_ultrasonic_sensor_code

closed 2016-08-17 by sant0ro