UIoT/middleware_sc_execl #5: Integração com modulos WiFi - ESP8266

felipeduarte34, 2016-04-16:


tomsqueiroz, 2016-04-16:
Tiraram o código de servidor TCP do git, eu envio de novo pra poder citar no comentário de finalização de tarefa ou só cito mesmo ?

tomsqueiroz, 2016-04-17:
O teste 1 já foi realizado. Utilizamos o módulo ESP8266 como cliente TCP para se conectar ao Raspberry Pi 2 que rodava  ArchLinux. O código utilizado para o arduino se encontra no GitHub: middleware_sc_execl/execl_use_cases_codes/new_codes/arduino/esp8266_tcp_client. O principal problema foi a não-estabilidade do ESP8266 devido a um erro no circuito. O erro foi resolvido alterando um divisor de corrente que possuía 1 resistor de 20kOhm e 1 resistor de 10kOhm, que baixava a tensão para um valor muito próximo da exigida pelo módulo e dificultava a comunicação entre o Arduino Uno e o ESP8266. A alteração consistiu em trocar o resistor de 20kOhm por outro de 10kOhm, o que implicou em uma tensão mais adequada para o circuito.

Bunoviske, 2016-05-12:
Código em do esp8266 para comandos AT em andamento. Ele foi dividido em três módulos seguindo orientação a objeto: conexão (manipula as ações de conexão do esp), dados (manipulação dos dados) e comando AT (envia comandos AT pela UART). Está sendo desenvolvido um gerador de comandos seguindo um protocolo e esse código trata o esp como um TCP/UDP client no modo STATION. Para tratá-lo como ACESS POINT tem que alterar o código. Deixarei essa parte para depois, quando for necessário implementar no projeto.

closed 2016-08-17 by sant0ro