UIoT/middleware_sc_execl #6: Getting more done in GitHub with ZenHub

felipeduarte34, 2016-04-16:
Hola! @felipeduarte34 has created a [ZenHub](http://www.zenhub.io) account for the **UIoT** organization. ZenHub is the only project management tool integrated natively in GitHub – created specifically for fast-moving, software-driven teams.

----

#### How do I use ZenHub?

To get set up with ZenHub, all you have to do is **[download the browser extension](https://www.zenhub.io?utm_source=ZHOnboarding)** and log in with your GitHub account. Once you do, you’ll get access to ZenHub’s complete feature-set immediately.

#### What can ZenHub do?

ZenHub adds a series of enhancements directly inside the GitHub UI:

- Real-time, customizable task boards for GitHub issues;
- Multi-Repository burndown charts, estimates, and velocity tracking based on GitHub Milestones;
- Personal to-do lists and task prioritization;
- Time-saving shortcuts – like a quick repo switcher, a “Move issue” button, and much more.

### [Add ZenHub to GitHub](https://www.zenhub.io?utm_source=ZHOnboarding)

_Still curious? See [more ZenHub features](https://www.zenhub.io/features?utm_source=ZHOnboarding) or read [user reviews](https://chrome.google.com/webstore/detail/zenhub-for-github/ogcgkffhplmphkaahpmffcafajaocjbd/reviews). This issue was written by your friendly ZenHub bot, posted by request from @felipeduarte34._

[image]

Bunoviske, 2016-04-21:
O pareamento entre dois dispositivos bluetooth já foi realizado, mas ainda é preciso testar o envio de dados. Para o modulo entrar em modo de configuração por meio de comandos at, é necessário seguir as etapas a seguir:
-ligue o arduino e retire o jumper de alimentação do modulo bluetooth
-segure o botão do modulo, e enquanto faz isso, ligue o modulo conectando
o jumper responsável pela alimentação.
-após ligar, basta soltar o botão e ele estará em modo de configuração

Lembra-se que o pareamento sempre deve ser feito entre um master e um slave.
Código:middleware_sc_execl/execl_use_cases_codes/new_codes/arduino/bluetooth-HC05/at_commands.ino

Bunoviske, 2016-04-23:
Troca de dados concluída. 

Código para o master:  middleware_sc_execl/execl_use_cases_codes/new_codes/arduino/bluetooth-HC05/master_senddata.ino 

Código para o slave:  middleware_sc_execl/execl_use_cases_codes/new_codes/arduino/bluetooth-HC05/slave_receivedata.ino

closed 2016-08-17 by sant0ro